platformio/platformio-docs #14: Here's a Gitlab build script

buckle2000, 2017-08-30:
So I just transpiled this from the Travis CI template in your doc.
https://gist.github.com/buckle2000/302e40aa0dfec81c64eebe01f14af806
Maybe it is useful.

ivankravets, 2017-08-30:
@valeros please extend docs with CI for GitLab

closed 2017-08-31 by valeros

buckle2000, 2017-08-31:
@valeros 
Still some problems in the docs.

## 1

[image]

## 2

Don't use `before_script` to install pip package!
`before_script` will run before **every** job. That's a waste of time.

## 3

You may mention that Gitlab has public linux runners, so most people don't need to setup one.

valeros, 2017-08-31:
@buckle2000 
1. It's ok, that reference will be resolved in the real doc page.
2. Have you tried your configuration for a real project example? I've tried using `job_install` as in your snippet, but as I saw from GitLab documentation for "jobs", they are totally independent, so PlatformIO won't be visible from other jobs. Am I missing something? 
3. Feel free to add any remarks as a PR.

buckle2000, 2017-08-31:
2. Sorry, I misunderstood something.

commit fcdb3d16 (2017-08-31): Add documentation for GitLab CI // Resolve #14