platformio/platformio-docs #33: Inconsistencies in "Arduino and Nordic nRF52-DK: debugging and unit testing" section

Navis-Raven, 2018-08-23:
Go to this link ("Writing Unit Tests" section): https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html

Then go to the unit testing section:
[image]

We remark two inconcistencies:
(1) :1234: It talks about "trhee" functions added while thre is four functions here and only two fuctions added to the main arduino functions "setup" and "loop".
(2) :1234: It says that setUp and tearDown function will be used, but they are never used in the example
(3) :1234: Other functions are used but never described or explained.

Navis-Raven, 2018-08-23:
This issue is linked to this one : https://github.com/platformio/platformio-core/issues/1805

Navis-Raven, 2018-08-23:
(7) Look at the picture below (tooken from the same documentation webpage) one of the two tests which are repeated 5 times should fail, while in the results all of thoose tests passed. 

[image]

In fact I think this is the reason: The output pit to BE an output pin, so it's likely that you're configuring the output as HIGH/LOW but it's always LOW because it's still being treated as the default (an INPUT). If this is the reason it has to be explained in the documentation otherwise we don't understand. 

BUT, If we see in the previous example at this webpage : https://docs.platformio.org/en/latest/tutorials/core/unit_testing_blink.html , the result are still PASSED (I've tested it with an ESP-32, and it works).

[image]

So I've put the same code but with the nRF52-DK this time and it failed, WHY ??? What the hell is going on ?

[image]

(8) Another inconsistency, the unity test module usage is wrong, the expected result should be before the "measure"

like in the example here (source: http://www.throwtheswitch.org/unity/)
[image]

If not doing like in the example overhead  when there is a failure the expected result is at the place of the measure and the measure at the place of the expected result. 

[image]

It shoud be "Expected 1 Was 0" and not the contrary.

Navis-Raven, 2018-08-23:
Now, At "adding Bluetooth LE capabilities" https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html#adding-bluetooth-le-features

There is two issues

(9) There is a minor issue here : There is a superfluous carriage return 
[image]

(10) At the end I think the documentation is not up to date, but I'm not sure. My smarpthone give this:

[image]

while the documentation watch this
[image]

ivankravets, 2018-08-24:
@valeros could you fix tutorial and help here?

valeros, 2018-08-27:
Hi @Navis-Raven! Many thanks for pointing out these issues.
As for the note `(7)`, indeed, there is a problem with function `digitalRead` since it always returns `LOW` despite the fact that `digitalWrite` was called with `HIGH` value (LED blinks so `digitalWrite` works as expected). We aren't responsible for the third-party software, so probably it would be better to open an issue in their official repository.

Navis-Raven, 2018-08-27:
No, pb
So, in that case you can put a nota bene for the user who read this tutorial, in order to not be troubled by this bizarre thing. 

Do you know where can we open this issue in their repository.

Navis-Raven, 2018-08-27:
This the message I'm trying to send to Nordic (but at this time, their website have an issue, lol, impossible to post anything):

There is a notable bug with nRF52-DK with arduino framwork, as you can see on github here (see the point 7) : https://github.com/platformio/platformio-docs/issues/33#issuecomment-416382889

The problematic code is here: https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html#writing-unit-tests

[image]

there is a problem with function digitalRead since it always returns LOW despite the fact that digitalWrite was called with HIGH value. 

IDE: VSCODE with platformio, 
SDK/Framwork: Arduino

Navis-Raven, 2018-08-28:
@valeros after the last documentation update I have to add some unclear points (question/issues/inconsistencies) to this tutorial: https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html

We are still wanting to use nRF52-DK board, now following this link for installing drivers for J-Link (on board debug tool in our case)
[image]

So qe've just openned this link ( https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#debugging-tool-jlink ) which describes J-Link external debug probe. Here we have to install J-Link drivers. 

(8) It says that we have to install J-Link GDB Server, but the documentation does not explain what is it nor say why we have to install it:
[image]

(9) When we want to install GDB server for nRF52-DK on board, it opens this webpage then we click on the button "downloads" 
[image]

But there is many links and we don't know which one to choose, and no one is called "J-Link GDB Server"

[image]

There is only one which is an RTOS plugin for Eclipse which does'nt match with what we want. 
[image]

So I skipped this step. 

(10) return back to this webpage: https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#debugging-tool-jlink

And go directly to driver section. We are on Windows (8.1) an we have to follow windows installation process
[image]

BUT, it says to start debugging session
(10.a) here we are still following the beginner tutorial, and to this moment, the user don't know how to start debugging session
(10.b) it cannot work since we didn't configure platformio.ini with `debug_tool = jlink`  
(10.c) and the beginner user does not know who to configure paltformio.ini. You can lpok at the tutorial webpage : https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html 

You can see that the only one ini configuration suggested is for adding BLE service and does not mention  anything about the debug tool
[image]
When not mentionned, I remarked that the debug tool choosen is CMSIS-DAP link automatically (which I don't recommend since it has installer issues with nRF52-DK at this time: https://github.com/ARMmbed/mbed-cli/issues/733

(10.d) Let's add `debug_tool = jlink` in the ini file now, start debug session, and see what happens : it works and it should'nt work because we didn't installed yet the driver. How it can be ?
(10.e) Our objective was to allow code debugging by installing the driver. So as we can now continue the tutorial without even installing the driver, we can ask the question: why installing it ? What is used for ? Why the documentation says to install a debugging driver afterstart  debugging I really don't understand that. 

NB: Note that I've previously  talked about the point 10.d and 10.e here: https://github.com/platformio/platformio-core/issues/1802

Navis-Raven, 2018-08-29:
For (9) a segger responsible told me that they will modify the website accordingly of my question "where is GDB Server ?". So Segger should solve this problem. However I learnt that the GDB Server is include inside the J Link software installer, so for the point (8) you can make more clear for the newcommer that we just have to install J-Link tool or anything else (I'm not suggesting anything because I still not understand the purpose)

Navis-Raven, 2018-08-30:
@valeros (UPDATE)
The nordic devzone website is working again, I've just left them a message with a brief issue description and a link to your coment. 

If you want to participate as a PlatformIO representative, it's here: https://devzone.nordicsemi.com/f/nordic-q-a/38028/there-is-a-notable-bug-in-nrf52-dk-with-arduino-framework

Navis-Raven, 2018-08-31:
(11): There is a new "test icon" so you can make the images of the documentation up to date, and its instructions 

[image]

[image]

Navis-Raven, 2018-08-31:
(13) The code example for unit testing does not compile

[image]

Navis-Raven, 2018-08-31:
(13): Why a delay has been added? Is it for avoiding unity masked results ? The documentation needs to explain:
[image]

ivankravets, 2018-08-31:
That was a typo when we missed this delay in the tutorial. It was added to docs before. See warning block here http://docs.platformio.org/en/latest/plus/unit-testing.html#workflow

Navis-Raven, 2018-08-31:
@ivankravets ok thanks for info but I think that the tutorial need to explain this, because the tutorial is before the web page where the warning block is.

[image]

[image]

Navis-Raven, 2018-08-31:
@ivankravets  only few comments in the example code could be sufficient

ivankravets, 2018-08-31:
@valeros will update this tutorial with new screenshots and fix all issues.

Navis-Raven, 2018-08-31:
(14): it's src not src_dir
[image]

(15) the same for STMCube tutorial
[image]

(16) You didn't remind to create "test" folder with "test_main.c" in it like you did with "main.c"

[image]

[image]

Navis-Raven, 2018-08-31:
Many issues in STM32Cube tutorial as well:

[image]

(17) same as (1)

(18) same as (14)

(19) should be #include "../src/main.h" or it wouldn't compile (I tested the compilation on VSCode.

(20) I've tried to make the STM32Cube tutorial on VSCode with  ST DISCO-L475VG-IOT01A and it was close (I made little modifications off headers off course "stm32l4xx_hal.h" instead of the other one), I don't know if it is a misuse, or a bug (documentation issue included). But the tests wasn't able to start since it was waiting for reset, even if I pushed the reset button it gave nothing, the test console was stuck at this step (see below)

```

> Executing task: C:\Users\charafeddine\.platformio\penv\Scripts\platformio.exe test <

PIO Plus (https://pioplus.com) v1.4.6
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

============================================================================================= [test/*] Building... (1/3) =============================================================================================
Please wait...

============================================================================================= [test/*] Uploading... (2/3) =============================================================================================
Please wait...

============================================================================================== [test/*] Testing... (3/3) ==============================================================================================
If you don't see any output for the first 10 secs, please reset board (press reset button)

```

Navis-Raven, 2018-08-31:
@ivankravets while the test was stuck in the console I was able to build a new code and upload at the same time, is this normal that two concurrent tasks runs at the same time ?
[image]

Navis-Raven, 2018-08-31:
(21): Dead Link
[image]

[image]

Navis-Raven, 2018-09-03:
@valeros can you answer to "jandrassy"'s question in this post ?
https://github.com/sandeepmistry/arduino-nRF5/issues/308

It's about the digitalRead issue

Navis-Raven, 2018-09-06:
@valeros 
Now we have the real story about the point (7) (https://github.com/sandeepmistry/arduino-nRF5/issues/308#issuecomment-418916739), fortunately or unfortunately it is an issue documentation as well

[image]

Navis-Raven, 2018-09-06:
@ivankravets good to hear but, I have to check all of the points first, it will take a little time, I'll inform you later.

Navis-Raven, 2018-09-06:
No don't close it I see some points which are not corrected yet, I'll make a recap later

Navis-Raven, 2018-09-06:
So the unsolved points are: 
6.
8.5
9.
10. / 10 a to e
11. (but at the end)
13. (commentaries for the delay in the code) 
14&15. check my previous comment. 
18. unknown
19.
22.

And I need to test the point 7. and 12.

Navis-Raven, 2018-09-07:
7. no more applicable
12. no more applicable 
because the code has changed. 

The new code is compiling and working well BUT, notice it is not related to the previous code, it does not appear as unit testing tutorial but a tutorial how to use unity.h with platformIO and this lead us the the point 23: 

[image]

A solution for this inconsistency:
So to correct this problem, in the section writting unit test  you can create previous paragraph (between writing unit test and "Test cases can be") accompanied by a source code that manages hello_world strings, And then you can explain "but as we cannot easily verify the content we need unit testing." And only after this you can say in the next paragraph: "we need to ADD four functions etc..." to test our code.
And then the source code which tests the strings. (the source code will be adapted to the previous).

Navis-Raven, 2018-09-07:
I've contacted segger they solved their website issue. 
So point 9 is solved
7 and 12 no more applicable

recap list of unsolved the points

6.
8.5
10. / 10 a to e
11. (to be done after GUI PlatformIO upgrades at the end)
13. (commentaries for the delay in the code)
14&15. check my previous comment.
20. unknown
21.
22.
23.

valeros, 2018-09-10:
`6` - should be fixed now
`8.5` - JLink is now a separate package, so the page about this tool should be rewritten
`10a-e` - `debug_tool` field has been added to the `platformio.ini` (partially fixed?)
`11` - should be fixed now
`13` - there is a `Note` section that describes the purpose of this delay
`14,15` - `src_dir` is just a link to the appropriate page so nothing to fix?
`20` - I don't have this board so can't reproduce
`21` - Link has been removed
`22` - `HAL_Init` is part of the initialization routine from the framework (nothing to comment?) and there is a `Note` section for about the delay
Anyway, some of the remarks described here don't belong to this issue, so it would be better to open separate issues for them?

Navis-Raven, 2018-09-10:
for `20` I'm investigating this with @ivankravets 
`14&15` ok
`22&13` In general it would be better to add comments (that can be small comments) in the code in order for a newcommer to identify the blocs relatively with the text and the note. For example `HAL Init`. 
The more the documentation is clear the less Ivan&PlatformIO have time to develop and fixing issue instead of answering same questions many times. 
`10` I still don't understand how can we launch the debug session successfully if the driver is not installed, can you explain me this ? https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#debugging-tool-jlink
For my case, I launched debug session and it worked without having to install `InstDrivers.exe`, strange isn't it ? 
`6, 11` waiting for the fix
`21` the link is still here. 

You forgot `23`

> Anyway, some of the remarks described here don't belong to this issue, so it would be better to open separate issues for them?

Yes but they are indirectly linked to nRF52-DK. For example the debugger J-Link installation is a part of this tutorial prerequisites. As well, the points about STM32Cube tutorial are similar "bugs". Yes `23`don't belong to this issue, it's another tutorial issue, but if the link will be removed in the next documentation version, it will be good.

valeros, 2018-09-10:
`21` should be fixed now
`6`.  Am I missing anything or it's fixed?
[image]
Same for `11`:
[image]

Navis-Raven, 2018-09-11:
@valeros 
(21) still here
[image]

(6) there is something wrong on your fix, look by yourself
[image]

(11) I've told you to wait until the new GUI to update all PIO/VSCode user interface screenshots, to update the documentation accordingly. The new GUI is here now, see by yourself, there is a new run task button:

[image]

ivankravets, 2018-09-25:
@Navis-Raven please file a new issue if something is not fixed.

closed 2018-09-25 by ivankravets

Navis-Raven, 2018-09-27:
ok

Navis-Raven, 2018-09-27:
I was sick these last weeks. I'll see that as soon as I recover my health

ivankravets, 2018-09-27:
Get well soon! 😊

Navis-Raven, 2018-10-02:
6: fixed
8.5: fixed
10.a: unsovled
10.b: fixed
10.c: fixed
10.d: it's a mystery
10.e: I still need an answer (I can open a separate issue for it)
11. some are fixed but not all of them the STM32Cube tutorial does not have up to date screenshots: https://docs.platformio.org/en/latest/tutorials/ststm32/stm32cube_debugging_unit_testing.html
20. I'm investigating with Ivan Kravets
21. fixed
22 fixed
23: unsolved.

(24) New one: The note : "2 sec delay is required since the board doesn’t support software resetting via Serial.DTR/RTS"
is missing on this ESP-32 tutorial, isn't it ? 
https://docs.platformio.org/en/latest/tutorials/espressif32/arduino_debugging_unit_testing.html

Navis-Raven, 2018-10-02:
@ivankravets @valeros  to summerize, these are the unsolved points :

10.a
11. some are fixed but not all of them the STM32Cube tutorial does not have up to date screenshots: https://docs.platformio.org/en/latest/tutorials/ststm32/stm32cube_debugging_unit_testing.html
23: unsolved.
(24) New one: The note : "2 sec delay is required since the board doesn’t support software resetting via Serial.DTR/RTS"
is missing on this ESP-32 tutorial, isn't it ?
https://docs.platformio.org/en/latest/tutorials/espressif32/arduino_debugging_unit_testing.html

for
10.d and 10.e I'll open a different issue for them
20. I'm inviestigating in PlatformIO community with Ivan. As soon I have the answer I'll open new issue or not (depending on the results)

ivankravets, 2018-10-03:
Please open new issue.

Navis-Raven, 2018-10-03:
Done here
https://github.com/platformio/platformio-docs/issues/43