platformio/platformio-docs #34: The tables in the platforms sections does not display completely

Navis-Raven, 2018-08-24:
The tables in the platforms sections does not display completely

For example for ESP32:
Like here at the page, at this link: https://docs.platformio.org/en/latest/platforms/espressif32.html#configuration

[image]

We see the problem particulary here for the STM 32
https://docs.platformio.org/en/latest/platforms/ststm32.html#id114

For example we want to see the continuation of this higlighted line: 
[image]

Then we click on scroll to the left, with the mouse at the bottom of the table
[image]

And now we try to return to see our line...but where is it ???
[image]

Navis-Raven, 2018-10-02:
Is there a functionnality to freeze the first column/line of the tables ? If there is, This will solve the problem

ivankravets, 2018-10-03:
Duplicate of https://github.com/platformio/platformio-docs/issues/5

Tables were refactored. Now, each board has own page with rich documentation.

closed 2018-10-03 by ivankravets