platformio/platformio-docs #36: Debug vs Debug (skip pre-debug)

Navis-Raven, 2018-08-28:
hi, 
there is two options in the debug menu: "PIO Debug" and "PIO Debug (Skip Pre-Debug)"

[image]

But I didn't find in the documentaion what is the difference between each of them. The only thing which I found about pre-debug in the documentation is about the settings (release vs pre-debug):

[image]

[image]

ivankravets, 2018-09-06:
See new docs: http://docs.platformio.org/en/latest/ide/vscode.html#debugging

1. Please upgrade PIO Core
2. Now, default debug configuration is without "Pre-Debug"
3. defaultToolbarAction setting was removed. Use http://docs.platformio.org/en/latest/projectconf/build_configurations.html

closed 2018-09-06 by ivankravets

Navis-Raven, 2018-09-06:
Problem solved
[image]

commit f66c27d2 (2018-09-06): Document "Debugging" for VSCode // Resolve #36