platformio/platformio-docs #38: Bad "next" link in platformIO documentation

Navis-Raven, 2018-08-29:
Hi,
Go to this webpage : https://docs.platformio.org/en/latest/userguide/account/cmd_token.html

Note that we are on "platformio account token" section
Then click on Next:
[image]

And after clicking on next you return back to the previous "platformio boards" section
[image]

So the "next" link is not correct

Navis-Raven, 2018-08-29:
I didn't check all previous and next buttons it's the first time I see this bug, however you shall check the rest of them if there is no other bad next/previous link like this one (by clicking "next" from the beginning to the "end", and after "previous" from the end to the beginning.

Navis-Raven, 2018-08-29:
*** UPDATE ***
I've seen rapidly some other inconcistencies like the "next" button jumps directly to the next chapter and not like example before, or the countrary (it does not jump som subchapter). So please check these "next" "prevous" button to be sure, it jumps the "jumpable" subchapters and do not jump important chapters.

Navis-Raven, 2018-10-02:
this one is still unsolved

pfeerick, 2019-05-21:
Hm... I think it's an unfortunate side effect of reusing already written documentation... 
```
CLI Guide
---------

.. toctree::
    :maxdepth: 2

    ../userguide/account/index
```

For want of another better way to resolve this, perhaps instead of just re-purposing the already existing page using a relative link, do a hard file link at the file system level instead? Preserving the navigation system layout, but not having maintain multiple copies of the file?