platformio/platformio-docs #39: espressif8266: OTA: Documentation doesn't mention it doesn't work with HttpUpdateServer

MihailMihov, 2018-09-08:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#over-the-air-ota-update
  http://docs.platformio.org/en/latest/platforms/espressif8266.html#authentication-and-upload-options

  You can't give a username/password to the upload command only a password (via -auth), which is how ArduinoOTA works. The [What is OTA? How to use it?](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html) links to a ESP8266 guide that explains about both methods (ArduinoOTA and HttpUpdateServer).
  Am I missing something and you actually can use HttpUpdateServer, or?

valeros, 2018-09-19:
We support only the `ArduinoOTA` method at the moment. But I suppose you should be able to interact with `HttpUpdateServer` using [extra_script](http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-upload-tool) and `curl`

pfeerick, 2019-05-21:
I suggest this issue can be closed. 

ArduinoOTA is for flashing with the Arduino IDE/PlatformIO, etc. ESP8266HTTPUpdateServer is for interactive _Go to the ESP8266's update-me web page and upload binary_ flashing ... which you should be able to use with PlatformIO if you wanted as you could build your project, and then grab the bin file from the project `.pioenvs` folder.  Finally, for ESPhttpUpdate, that's supposed to run on the ESP8266, and check a configured server for updates at whatever interval or time you set.

closed 2019-05-21 by MihailMihov