platformio/platformio-docs #42: Questions about J Link installation documentation

Navis-Raven, 2018-10-02:
Go to this webpage: https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#debugging-tool-jlink

FIRST QUESTION: Let's add debug_tool = jlink in the ini file now, start debug session, and see what happens : it works while it should'nt work because we didn't installed yet the driver. How it can be ? please explain.

SECOND QUESTION: This question depends of the answer of the first question.

ivankravets, 2018-10-03:
See https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#drivers

When you start debugging session, PlatformIO downloads depending `tool-jlink` package into `~/packages/tool-jlink`. After that, you can navigate here and install driver.

closed 2018-10-03 by ivankravets

Navis-Raven, 2018-10-03:
@ivankravets  but it's a mystery that the debug session worked without installign this downloaded driver (and I didn't even installed it before).