platformio/platformio-docs #65: ATMEGA32U4 boards and ModemManager on Linux

pfeerick, 2019-07-02:
So, it appears there is an issue in Debian Buster and Ubuntu 18.04.2 already, and more Linux distros will follow suit due to a change in behaviour in the ModemManager package exposing a bug in the firmware used on ATMEGA32U4 USB bootloaders, so boards like the Arduino Micro and Sparkfun Pro Micro are affected, just to name two... would adding something to the FAQ be more appropriate than to each boards page?

I found it mentioned here after banging my head against errors for a good half hour, and trying to craft udev rules to stop it, to no avail: Arduino/issues/7690

---

`error: programmer did not respond to command: set addr` - Newer versions of ModemManager are known to cause this error when trying to upload to your Arduino Micro or Sparkfun Pro Micro if trying to use the built in USB bootloader. The upload is prone to failure, corrupting the upload, and may require you to use an In Circuit Serial Programmer in order to re-write the device bootloader. 

A temporary fix is to run `sudo systemctl stop ModemManager` from a terminal prompt, which will last until you reboot your computer. The more permanent solution is to run `sudo systemctl disable ModemManager`. Unless you use mobile broadband you may not even notice it's disabled.

dlarue, 2019-07-15:
Just chiming in to say I've been battling this issue on Kubuntu 18.04.2 first with Arduino 1.8.8 and then 1.8.9 and purging ModemManager didn't change anything.  The ATMEGA32U4 boards I'm attempting to program are the Pro Micro Beetle inwhich the bootloader says it's a GenuinoMicro.

I had deleted my ~/.arduino15 directory upon installing Arduino IDE 1.8.9 which didn't help but strangely, when I moved that out of the way and replaced the old ~/.arduino15 directory I was able to get one upload to complete. Dozens of attempts afterwards would not work.

pfeerick, 2019-07-16:
I'll spin up a Kubuntu 18.04.2 VM with Arduino 1.8.9 tonight and see if I can reproduce this... fingers crossed ;) The board I have is a Arduino Micro clone (5v), which has the official Arduino bootloader, so hopefully that doesn't exhibit different behavior

dlarue, 2019-07-16:
Thanks, I'll boot a custom Lubuntu 18.04.2 iso and try it there too.  The board I'm using is the 5V, ACROBOTIC PRO Micro Beetle ATmega32u4 Development Board Arduino-Compatible Nano Micro | USB-A
https://www.amazon.com/gp/product/B07P8Y3TS8

pfeerick, 2019-07-16:
Sorry, couldn't reproduce that on my end. The board I'm using is one of these...

https://www.ebay.com/itm/New-Pro-Micro-ATmega32U4-5V-16MHz-Replace-ATmega328-Arduino-Pro-Mini-/221891843710

I didn't even kill modemmanager on the Kubuntu VM... it didn't seem to cause any issues yet... although that could be something to do with the VM as well. I also didn't install the udev rules, which I did expect to need to do. I did add the user to the `dialout` group though (it was already in the plugdev group, so I didn't do that one), and logged out and in again to apply the group permission changes. After reloading the bootloader over ISCP as it was being stupid, I had no problems with programming a plain led blinky program blinking an led connected to D2.

dlarue, 2019-07-16:
Thanks, I had tried to save you some time and posted updates to my original posting stating that I got PlatformIO to work. I too didn't change the plugdev rules but did remove ModemManager and it worked like a charm.

pfeerick, 2019-07-16:
Thanks for that ;) I wasn't sure if it was a one off, or was consistently working, and also gave me a chance to try the 1.8 beta of the vscode extension on another linux variant ;) At least you've been able to get it working, that's the important thing! :)

dlarue, 2019-07-17:
Yes, getting the software uploaded and with a platform I can work out the software in was the goal.  And with that the solution now has me dipping my toes into the PlatformIO waters. I had originally thought this post was another Arduino git issue.  Gotta love the Internet and how things can lead from one thing to another. Thanks for the guidance.

pfeerick, 2019-08-31:
I've since realised it should be possible to simply add a udev rule to instruct modem-manager to stop being a PITA, so will close this issue as I have done a PR against the udev rules which would negate this issue entirely. https://github.com/platformio/platformio-core/pull/2966

closed 2019-08-31 by pfeerick