platformio/platformio-docs #70: lib_deps_builtin outdated?

Rotzbua, 2019-07-19:
The snippet of platformio.ini rises an error. Is it outdated?

#### Snippet

```
lib_deps_builtin =
    SPI
    Wire
```
Found on: https://docs.platformio.org/en/stable/projectconf.html

#### Steps to Reproduce

1. run `platformio.exe init`
2. Shows: `Warning! Ignore unknown configuration option `lib_deps_builtin` in section [env:nodemcu-32s]`

ivankravets, 2019-07-19:
It’s declared in custom section, not env. You can use any name for option in custom sections.

lemonkey, 2022-03-13:
FYI the documentation clearly shows an example where "lib_deps_builtin" is being defined within the [env] section: https://docs.platformio.org/en/latest/projectconf/index.html

udem1234, 2023-01-23:
@ivankravets : I just realized the same (that the documentation clearly shows an example where "lib_deps_builtin" is being defined within the [env] section).  Is this a mistake ?

ivankravets, 2023-01-23:
No, it is not a mistake. You can declare an internal library from a framework as your project dependency using the `lib_deps` option. It can be used for SPI and other built-in libraries.

ardnew, 2023-09-11:
I don't agree with this getting closed. 

If the example[^1] is **_incorrect_**, then the examples need to be updated to demonstrate a valid, **_correct_** configuration file. 

However, if the example _somehow_ is **_correct_**, then running[^2] that example should **not** generate warnings. 

In either case, a change should have been made originally and I believe still does.

[^1]: PlatformIO online documentation lists the following [example configuration file](https://docs.platformio.org/en/stable/projectconf/index.html):
    ```ini
          ┆  [platformio]
          ┆  default_envs = nodemcuv2
          ┆  
          ┆  ; Set/override default options for each "[env:***]"
    »HERE»┆  [env]
          ┆  build_flags =
          ┆      -D VERSION=1.2.3
          ┆      -D DEBUG=1
    »HERE»┆  lib_deps_builtin =
          ┆      SPI
          ┆      Wire
    »HERE»┆  lib_deps_external =
          ┆      bblanchon/ArduinoJson @ ~5.6,!=5.4
          ┆      https://github.com/gioblu/PJON.git#v2.0
          ┆      IRremoteESP8266=https://github.com/markszabo/IRremoteESP8266/archive/master.zip
          ┆  
          ┆  [env:nodemcuv2]
          ┆  platform = espressif8266
          ┆  framework = arduino
          ┆  board = nodemcuv2
          ┆  
          ┆  ; Build options
          ┆  build_flags =
          ┆      ${env.build_flags}
          ┆      -DSSID_NAME=HELLO
          ┆      -DSSID_PASWORD=WORLD
          ┆  
          ┆  ; Library options
          ┆  lib_deps =
          ┆      ${env.lib_deps_builtin}
          ┆      ${env.lib_deps_external}
          ┆      https://github.com/me-no-dev/ESPAsyncTCP.git
          ┆      knolleary/PubSubClient@^2.8
          ┆      paulstoffregen/OneWire
          ┆  
          ┆  ; Serial Monitor options
          ┆  monitor_speed = 115200
          ┆  monitor_flags =
          ┆      --encoding
          ┆      hexlify
          ┆  
          ┆  ; Unit Testing options
          ┆  test_ignore = test_desktop
          ┆  
          ┆  [env:bluepill_f103c8]
          ┆  platform = ststm32
          ┆  framework = arduino
          ┆  board = bluepill_f103c8
          ┆  
          ┆  ; Library options
          ┆  lib_deps = ${env.lib_deps_external}
          ┆  
          ┆  ; Debug options
          ┆  debug_tool = custom
          ┆  debug_server =
          ┆      ${platformio.packages_dir}/tool-jlink/JLinkGDBServer
          ┆      -singlerun
          ┆      -if
          ┆      SWD
          ┆      -select
          ┆      USB
          ┆      -port
          ┆      2331
          ┆      -device
          ┆      STM32F103C8
          ┆  
          ┆  ; Unit Testing options
          ┆  test_ignore = test_desktop
    ```

[^2]: Running a project based on the example configuration file above (from the online documentation) generates unexpected warnings:
    ```sh
    $ pio run -t upload --upload-port /dev/ttyUSB0
    Warning! Ignore unknown configuration option `lib_deps_builtin` in section [env]
    Warning! Ignore unknown configuration option `lib_deps_external` in section [env]
    ```

ivankravets, 2023-09-13:
Thanks for the report, fixed!

commit 325ff802 (2023-09-13): Fix example with the custom common options // Resolve #70
closed 2023-09-13 by ivankravets