platformio/platformio-docs #98: Needs Framework=Arduino to compile

gmawdsley1, 2019-11-22:
[env:d1]
platform = espressif8266
framework = arduino
board = d1

ivankravets, 2019-11-22:
where?

gmawdsley1, 2019-11-23:
Trying to compile an example for Wemos D1 R1. Compilation was blocked until I added framework. The tutorial did not mention that. I also can’t upload because there is no port.

This e-mail is intended only for the named recipient(s) and may contain confidential, personal and/or health information (information which may be subject to legal restrictions on use, retention and/or disclosure).  No waiver of confidence is intended by virtue of communication via the internet.  Any review or distribution by anyone other than the person(s) for whom it was originally intended is strictly prohibited.  If you have received this e-mail in error, please contact the sender and destroy all copies.

ivankravets, 2019-11-25:
So, could you provide a link?

pfeerick, 2020-02-13:
I would suggest this issue can be closed unless further information / a link is provided. The only places I see the `[env:d1]` mentioned in the docs are in relation to the board configuration. The nearest match I could find is ...

```
[env:d1]
platform = espressif8266
board = d1
```
... from the d1 board info page (https://github.com/platformio/platformio-docs/blob/develop/boards/espressif8266/d1.rst#configuration) and that is not a tutorial or an working snippet anyway, as it's still up to the end user to choose the framework they wish to use after configuring the board type.

closed 2020-02-13 by ivankravets