platformio/platformio-docs #112: inquiry about articles

HamzaHajeir, 2020-01-27:
Hello Platformio-ers 

I'm writing about PlatformIO in a **series of articles**.

My way is to publish first at a facebook group, to be prepared to publish a crafted version in **LinkedIn**.

### I've 2 Questions in mind :
- Is it welcomed in Articles page to include articles written in other languages ? In my state I'm writing in Arabic.

- The series began with Arduino IDE as Open Source , travelling to PIO in the next 3 parts of the article, Is that welcomed ?

### And here's the bonus question (in case of OK response to previous 2 Q's) : 
- Should I add them in **one line** as one article, or it's accepted in **4 separate lines** ?

Thanks in advance (y)

ivankravets, 2020-01-27:
Hey,

We would be thankful for any materials about PlatformIO. We will add later reference to your article in our docs. How about to publish them on Medium or Hackster.io?

HamzaHajeir, 2020-01-27:
Thanks for your suggestions, 

The main obstacle here is the language I wrote them with. I've targeted Arabic audience. So I've doubts the article wont be welcomed in such sites.

HamzaHajeir, 2020-11-11:
Hi

I've already published all parts in Arabic to my LinkedIn Profile, If it's welcomed and you're satisfied, You can address them in "Articles about PlatformIO" section :

[PlatformIO Part 1 : The Open Source Initiatives
](https://www.linkedin.com/pulse/platformio-part-1-open-source-initiatives-hamzah-hajeir/)
[PlatformIO Part 2 : The Professional Open-Source
](https://www.linkedin.com/pulse/platformio-part-2-professional-open-source-hamzah-hajeir/)
[PlatformIO Part 3 : Real Advancement
](https://www.linkedin.com/pulse/platformio-part-3-real-advancement-hamzah-hajeir/)
[PlatformIO Part 4 : Ultimate features
](https://www.linkedin.com/pulse/platformio-part-4-ultimate-features-hamzah-hajeir/)

My pleasure

ivankravets, 2020-11-11:
Thank you so much! The new docs will be published in 5-10 minutes! 😊

commit a3209b7f (2020-11-11): Add articles by Hamzah Hajeir // Resolve #112
closed 2020-11-11 by ivankravets