platformio/platformio-docs #124: Creating new project in CLion doesn't work

gbaranski, 2020-03-30:
CLion version 2019.3.5 x64, doesn't work on both Windows 10 and Mac os catalina, shows "Gathering info..." and on bottom "PlatformIO utility is not found". I can't find any fix
[image]

ivankravets, 2020-03-31:
This is a known issue, we requested this feature here  https://youtrack.jetbrains.com/issue/CPP-19412
CLion does not install PlatformIO Core automatically for you:

1. [Install PlatformIO Core](https://docs.platformio.org/en/latest/core/installation.html#installation-methods) using installation script
2. [Install Shell Comamnds](https://docs.platformio.org/en/latest/core/installation.html#install-shell-commands)

So, you should have working `platformio --version` command when opening the system terminal or after the restart of your OS. You need to patch the PATH system environment variable and extend it with a folder where PlatformIO Core is installed. We don't have SUDO/Administrative permissions. As a result, we can't do this for you.

The other option is to install [PlatformIO IDE for VSCode](https://docs.platformio.org/en/latest/integration/ide/vscode.html), it will install PlatformIO Core automatically for you. You will just need to path the system PATH and switch back to CLion.

/cc @anastasiak2512 @elmot

gbaranski, 2020-03-31:
Awesome, thanks :)

closed 2020-03-31 by gbaranski

ivankravets, 2020-03-31:
Does it work now?

gbaranski, 2020-03-31:
@ivankravets that works, now I'm trying to figure how to fix syntax highlithing, CLion doesn't see default Arduino.h library

elmot, 2020-03-31:
@gbaranski ensure that your files are c++ (i.e. have .cpp extension)

Sukikui, 2023-07-06:
> 

Hello, can you more precisely explain how to "path the system PATH" ? I met the same problem and nothing work, but I've PIO Core thanks to VSCode, from this I don't know how to link the packages with CLion.

VA7ODR, 2024-10-27:
I also encountered the issue where it always says "Gathering info..." forever if you try to create a new PlatformIO project while currently editing another.  If you close all extra instances, and switch the remaining CLion instance to a Non-PlatformIO project, then create the new one, it instantly shows the board list.

biryusal, 2024-11-13:
> I also encountered the issue where it always says "Gathering info..." forever if you try to create a new PlatformIO project while currently editing another. If you close all extra instances, and [...]

I just don't know what to say...
You saved me days of work!

I really had to close the previous project (that was also created via `PlatformIO`) and then the endless `Gathering info` has gone

thank you so much!

> I also encountered the issue where it always says "Gathering info..." forever if you try to create a new PlatformIO project while currently editing another. If you close all extra instances, and [...]

> I also encountered the issue where it always says "Gathering info..." forever if you try to create a new PlatformIO project while currently editing another. If you close all extra instances, and [...]

I just don't know what to say...
You saved me days of work!

I really had to close the previous project (that was also created via `PlatformIO`) and then the endless `Gathering info` has gone

thank you so much!