platformio/platformio-docs #137: Clarify language on FAQ: Convert .ino to C++ files "Manually"?

LongLiveCHIEF, 2020-05-29:
In the faq, there is a section describing common changes you may need to make to `.ino` files to make them work as `.cpp`.

The header of this section uses the word "Manually" at the end, which implies that there is an automated way of doing this somewhere.

This can cause confusion (see https://github.com/Bodmer/TFT_eSPI/pull/637)

I haven't found anywhere in the docs where there is automatic handling of these by platformio prior to running the compile. 

I think the language can be cleared up a little to reduce confusion.  

I'd be happy to submit a PR with suggested language, if someone can confirm for me that there is not any automation to make .ino files compatible with .cpp built into platformio?

The first sentence of that document also does not make sense. I'm not sure what it's trying to say or imply by `Some Cloud & Desktop IDE doesn’t support Arduino files`

This again reinforces the implication that platformio _does_ handle this... but then why does it go on to describe the changes you need to make?

ivankravets, 2020-05-30:
We do an automatic translation of INO file to valid C++ file if you use INO file. INO is not a valid C++ format. It was invented for children to do quick prototyping. Arduino IDE automatically converts INO->CPP in background. You can't compile or treat INO as C++ file. This is why IDEs and compilers do not work with INO files.

ajlennon, 2021-07-17:
@ivankravets I am having a problem I can't nail down but am wondering if it is related to .ino -> ino.cpp conversion.

I have a load of legacy code with .ino files in. I am defining macros on the command line (actually platformio.ini) with build_flags and -D. However .ino files are failing to compile telling me the macros aren't defined.

Does this ring any bells for you at all?

Thanks

valeros, 2021-07-23:
@ajlennon It's hard to guess, but the .ino conversion shouldn't be a problem. It should be easily testable if you isolate one file that fails to compile and convert it manually to C++ just to see if the compiler still complains.

ajlennon, 2021-07-23:
Yeah I did that and I couldn't replicate it or I would have provided a test case. I still think the problem is in and around that area though

ajlennon, 2021-07-23:
I reckon if you git clone Tasmota and use a macro in an .ino which you defined on the command line then it'll fail.

ajlennon, 2021-07-24:
@valeros I did a but more looking into this to create a test case and the upshot is a I am an idiot. It works fine. Sorry to have bothered you!