platformio/platformio-docs #150: Debuger for nanoatmega328

porlock, 2020-09-14:
I found information on
https://docs.platformio.org/en/latest/boards/atmelavr/nanoatmega328.html#board-atmelavr-nanoatmega328

that " Arduino Nano ATmega328 has on-board debug probe and IS READY for debugging. You don’t need to use/buy external debug probe."

but only Compatible Tools is [simavr] that is a simulator, so it can debug but only on simulated env not actual board. I didn't find any way to debug code on Arduino Nano by plafrotmio. This information in documentation is misleading or not fully described.

pfeerick, 2020-09-16:
Pretty sure this *is* a documentation bug. I would not put simavr in the same category as a debugging tool/native debugging since it is only a simulator, and not necessarily a 100% or 1:1 with the actual AVR hardware. There is however discussion on the PlatformIO forum from around the same time simavr was implemented [of a library that could be used to allow native debugging to happen](https://community.platformio.org/t/another-tool-for-debugging-avr-microcontrollers/13776?u=pfeerick). 

I beleive since then Mauro managed to get all the changes into the avr-debugger library, so everything you should need to get started with it should be in that libraries repo. => https://github.com/jdolinay/avr_debug

ivankravets, 2020-09-16:
Sorry, what is wrong with docs? Debugging and simulation are totally different processes. Debugging requires hardware which could be simulated. The same as the process with drinking. You can drink water but also can drink wine. And the question, is something wrong with "drinking"?

porlock, 2020-09-17:
> Sorry, what is wrong with docs? 

Bellow statements  are miss leading,

**Arduino Nano ATmega328 has on-board debug probe and IS READY for debugging. You don’t need to use/buy external debug probe.**

Question is where is that onboard probe and how to use it ? 

Like you noticed debugging and simulation are totally different processes.
We can't just simply made a equal sign between debugging on physical board with on-board probe and debugging on simulated env.

ivankravets, 2020-09-18:
Ah, yes, now I understand. We will think about how to fix this.

dlschmidt, 2023-01-04:
Can you please finally update the documentation?
Then I would not waste multiple hours to get gdb running by compiling from source, only to find out that the documentation is _very_ misleading.

ivankravets, 2023-01-05:
Sorry for the issue, we have finally fixed in https://github.com/platformio/platform-atmelavr/commit/a6eafeb1a4c2be0d2ae839d2052d711c4878ba15

Docs will be regenerated soon.

commit 849a5401 (2023-01-05): Remove simulators from "onboard debug" list // Resolve #150
closed 2023-01-05 by ivankravets