platformio/platformio-docs #159: Oudated documentation about setting strict_ssl

Rotzbua, 2020-11-04:
## Problem

Command fails.

```bash
$ platformio settings set strict_ssl yes
Error: Invalid setting with the name 'strict_ssl'
The command "platformio settings set strict_ssl yes" failed and exited with 1 during .
```

**Change or removal is NOT mentioned in documentation, migration or release notes!!**

https://docs.platformio.org/en/latest/core/history.html
https://docs.platformio.org/en/latest/core/migration.html

## Solution

Update documentation to new? setting name.

Mention rename or removal in release note....

## Reference
https://github.com/platformio/platformio-docs/blob/develop/core/userguide/cmd_settings.rst#strict-ssl

ivankravets, 2020-11-04:
Sorry, fixed in https://github.com/platformio/platformio-docs/commit/b701762bef10ab52032b32e6721a7675534cb15f

closed 2020-11-04 by ivankravets

Rotzbua, 2020-11-04:
Thanks for clarification👍

ivankravets, 2020-11-04:
Happy coding with PlatformIO!

commit b701762b (2020-11-04): Document "Introducing Strict SSL/TLS" in migration // Resolve #159