platformio/platformio-docs #176: Suggestion: Allow Python 2.7 in unsupported mode

pitosalas, 2021-02-10:
**This may be a terrible idea,** but I just updated platformIO and was kind of tripped up by needing to install Python 3.7 to allow me to simply update my teensy. If the code still works with python2.x but you don't want to support it, it would be nice, as long as feasible to allow me to take the risk and still run it on 2.x, for example with a switch "--unsupportedpython

ivankravets, 2021-02-10:
What is your OS? Do you use PlatlformIO IDE for VSCode?

pitosalas, 2021-02-10:
I'm on Ubuntu. And I am just using the command line, and in fact am struggling a little because the PlatformIO cli keeps saying I have the wrong python even though I now have installed 3.6. It's a path problem which I am sure I can solve but really all I want to do is to refresh my teensy. The command I am issuing is 

$ platformio run --target upload

And I am getting: 

Error: Python 3.6 or later is required for this operation.
Please check a migration guide:
https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

Pito Salas
Faculty, Computer Science
Brandeis University

ivankravets, 2021-02-10:
Please perform 3 steps and the issue will be resolved:

- Remove existing PlatformIO Core https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system
- Check that you have this venv package https://github.com/platformio/platformio-core-installer/issues/85
- Install PlatformIO Core using official installer https://docs.platformio.org/en/latest/core/installation.html#installer-script

pitosalas, 2021-02-10:
> 	• Check that you have this venv package platformio/platformio-core-installer#85

I followed those steps and it is still giving the error message:

echo $PATH
/home/ubuntu/.platformio/penv/bin:/home/ubuntu/bin:/home/ubuntu/.local/bin:/home/ubuntu/.platformio/penv/bin:/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
[PI]~/linorobot_ws/src/linorobot/teensy/firmware$ python --version
Python 2.7.12
[PI]~/linorobot_ws/src/linorobot/teensy/firmware$ python3 --version
Python 3.6.8
[PI]~/linorobot_ws/src/linorobot/teensy/firmware$
[PI]~/linorobot_ws/src/linorobot/teensy/firmware$ platformio run --target upload
Error: Python 3.6 or later is required for this operation.
Please check a migration guide:
https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5
[PI]~/linorobot_ws/src/linorobot/teensy/firmware$

Pito Salas
Faculty, Computer Science
Brandeis University

ivankravets, 2021-02-11:
2-nd step. `sudo apt-get install python3-venv`?

then

```

pitosalas, 2021-02-13:
Thanks, I got it to work. Unfortunate that they decided to de-support python2.7 because I think many in the ROS robotics work are on that!

closed 2021-02-13 by pitosalas