platformio/platformio-docs #184: Proposal link for new library

aelray, 2021-03-10:
Proposal link for new library leads to archived [Github repository issue tracker](https://github.com/platformio/platformio-libmirror/issues).
[image]

ivankravets, 2021-03-14:
Thanks, fixed!

aelray, 2021-03-14:
Nice. But not actually the kind of fix that I was looking for. Is there some kind of documentation or support to wrap something which is already available as a PIO library? I am using [ERPC](https://github.com/EmbeddedRPC/erpc) in a legacy zephyr project and wanted to make it available as PIO library to move the whole project to PIO.

ivankravets, 2021-03-14:
Please file a feature request at https://github.com/EmbeddedRPC/erpc/issues

closed 2021-03-14 by ivankravets