platformio/platformio-docs #199: Documentation availability in PDF format

mickpf, 2021-05-05:
Dear Sirs,

I'm starting with platformio and missing offline documentation.

Is the "platformio-docs" available in PDF format? The PDF generation on "https://docs.platformio.org"  is turned off!

Kind Regards,
Michael

mickpf, 2021-06-03:
Dear Sirs,

It would be nice if someone could answer this question because my computer is disconnected from the Internet most of the time.
Or is there some way how to convert all documentation from GitHub to PDF? So far, I don't know any way.

Thanks in advance,
Michael

valeros, 2021-06-03:
Hi Michael ! Indeed there is no offline options at the moment, but it's possible to generate a local copy of the documentation manually. Anyway, I can write you here instructions on how to build it locally on your machine or just send you PlatformIO docs as a PDF/EPUB/Single HTML file. Let me know what's better for you.

mickpf, 2021-06-03:
Hello Valerii,

it would be very nice, if you could tell my how to convert the (whole) "platformio-docs" cloned from GitHub to a PDF under Ubuntu-20.04. Since some months I'm searching for a way, because I'd like to do the same thing with some other docs in the same format!

I'm seating in deep hole, where sometimes the WLAN doesn't work. So I always rely on the offline documentation.

Thanks in advance,
Michael

valeros, 2021-06-03:
Unfortunately, there is no easy way to generate a PDF file, but you can convert EPUB to PDF, or even use local HTML version. First of all, make sure you have Python 3, pip and `make` installed in your system. Here are the next steps:

```bash
# Install the "tox" package
pip install tox
# Clone PlatformIO Core sources with submodules
git clone --recursive https://github.com/platformio/platformio-core.git
# Navigate to the PlatformIO Core folder
cd platformio-core
# Create a venv for docs (it may start generating right away, feel free to cancel)
tox -e docs
# Activate docs env
source .tox/docs/bin/activate
# Navigate to the "docs" folder
cd docs
# Check available formats
make help
# Run specific target
make epub
```

closed 2021-06-03 by valeros

ivankravets, 2021-06-03:
I think the easiest way for offline documentation is to download multi HTML docs by https://github.com/platformio/platformio-docs/archive/refs/heads/gh-pages.zip

1. Please unpack ZIP archive
2. Open `en/latest/index.html`.