platformio/platformio-docs #217: Expose `--skip-packages` option in GUI

maxgerhardt, 2021-11-07:
The PIO Home -> Inspect screen should have an option that makes it so that `--skip-packages` is passed to `pio check`, otherwise ESP32, ESP8266 and SAMD based projects (and possible others) cannot be code-inspected in the UI because `cppcheck` fails.

See https://github.com/platformio/platformio-core/issues/3951 and https://community.platformio.org/t/code-inspection-not-returning-expected-result/21480.

ivankravets, 2021-11-07:
Sorry, it is our bug. We will update our docs.

closed 2021-11-08 by valeros

jerabaul29, 2021-12-14:
Same problem with Sparkfun Artemis using the core from nigelb.

GuyZou, 2022-01-05:
Please, Fix-it ...

maxgerhardt, 2022-01-05:
See https://github.com/platformio/platformio-core/issues/3951#issuecomment-993968462 for a `platformio.ini`-side fix, this should work with the GUI too.

commit 5a7258e0 (2021-11-08): Add documentation for the "skip-packages" option

Resolve #217