platformio/platformio-docs #230: Documentation bug - board_build.f_cpu

NormanDunbar, 2022-01-22:
In the docs for `board_build.f_cpu` at [Platform options — PlatformIO latest documentation](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html?highlight=board_build%20f_cpu#board-build-f-cpu)  the wording gives the impression that this will “overclock” the CPU:

> You can overclock a board by specifying a board_build.f_cpu value other than the default.

I would suggest changing this wording as that setting should only be used when the board has a different clock frequency to the default. Perhaps to:

> If you have changed a board’s clock frequency, then you should specify a board_build.f_cpu value to match the change.

your Leonardo board’s details are at [Arduino Leonardo — PlatformIO latest documentation](https://docs.platformio.org/en/latest/boards/atmelavr/leonardo.html#configuration), where again, it does rather give the impression that the ini file setting will overclock the board!

There's a discussion on the Community Forums at https://community.platformio.org/t/documentation-bug/25595 should you be interested in other opinions.

Cheers,
Norm.

valeros, 2022-01-24:
Hi @NormanDunbar ! Thanks for bringing it up. I've reformatted the description for that option, so hopefully, it's clearer now. The updated version will be available once the main PlatformIO repository syncrhonizes its docs submodule.

closed 2022-01-24 by valeros

NormanDunbar, 2022-01-28:
Hi @valeros,

thank you very much.

Regards,
Norman.

commit c8148aeb (2022-01-24): Clarify the usage of "board_build.f_cpu" option

Resolve #230