platformio/platformio-docs #241: Documentation re platformIO in Codeanywhere is obsolete and needs an update

Theremingenieur, 2022-05-02:
The documentation [https://github.com/platformio/platformio-docs/blob/develop/integration/ide/codeanywhere.rst](https://github.com/platformio/platformio-docs/blob/develop/integration/ide/codeanywhere.rst) is obviously obsolete. When connecting to Codeanywhere, the screens and container options are totally different, now. Obviously, there have been changes on the Codeanywhere side. Unfortunately, even when creating successfully a Python/Ubuntu container, step 4 of the tutorial 
`sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)" ` fails with an error message and it is thus impossible to run platformio on Codeanywhere.

ivankravets, 2022-05-03:
Could you provide PR?

Theremingenieur, 2022-05-03:
As soon as I get it working, I'll provide a pull request.

ivankravets, 2022-05-03:
> sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"

Could you try just `pip install platformio`? Without `sudo`, etc.

Theremingenieur, 2022-05-06:
Thank you Ivan, that works, but with a small quirk: A new Python Container on CodeAnywhere comes with the outdated pip version 20.2.3. Thus, I needed to run `pip install --upgrade pip` first, and afterwards, I could install PlatformIO with `pip install platformio`.

Theremingenieur, 2022-05-06:
Ah ok, thank you! Will this be improved one day?

ivankravets, 2022-05-06:
Please subscribe to this issue https://github.com/platformio/platformio-core/issues/3681