platformio/platformio-docs #255: Incorrect example in Googletest usage docs

Michal123456747, 2022-07-08:
In embedded tests, example line, that starts all tests should be without arguments.
https://github.com/platformio/platformio-docs/blob/develop/advanced/unit-testing/frameworks/googletest.rst

ivankravets, 2022-07-08:
Do you mean this https://github.com/google/googletest/blob/bf66935e07825318ae519675d73d0f3e313b3ec6/googlemock/test/gmock_output_test_.cc#L279 ?

Michal123456747, 2022-07-08:
No. In the link above is 
```
::testing::InitGoogleTest();
    // if you plan to use GMock, replace the line above with
    // ::testing::InitGoogleMock(&argc, argv);
```

instead of

```
::testing::InitGoogleTest();
    // if you plan to use GMock, replace the line above with
    // ::testing::InitGoogleMock();
```

ivankravets, 2022-07-08:
Thanks for the report! Fixed!

commit c72ff5c9 (2022-07-11): Fx "InitGoogleMock" for embedded target // Resolve #255
closed 2022-07-08 by ivankravets