platformio/platformio-docs #257: monitor_flags not available

ve-dima, 2022-07-10:
> Warning! Ignore unknown configuration option `monitor_flags` in section [env:328base]
`monitor_flags = 
	--echo
	--encoding
	cp1251`

ivankravets, 2022-07-10:
PlatformIO Core 6.1 :

- https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html

ve-dima, 2022-07-10:
Well, but without `monitor_flags`  i can't change text encoding. 

> --encoding[¶](https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmdoption-pio-device-monitor-encoding)
Set the encoding for the serial port (e.g. hexlify, Latin1, UTF-8), default UTF-8.

Does not support customization, like `echo`, for example

> --echo[¶](https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmdoption-pio-device-monitor-echo)
Enable local echo.
Can be customized in [“platformio.ini” (Project Configuration File)](https://docs.platformio.org/en/latest/projectconf/index.html#projectconf) using [monitor_echo](https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#projectconf-monitor-echo) option.

ivankravets, 2022-07-10:
Cp1251 - did save the source file in utf 8?

ve-dima, 2022-07-10:
Files in UTF-8, but is it really important in this situation?

Previously, using the `monitor_flags`, I could set a different encoding.
Now there is no `monitor_flags`, and you have to use the dialog (Ctrl + T + Ctrl + A) every restart of the Serial Port, configure everything anew.