platformio/platformio-docs #259: Nano 33 IOT libraries

jechiasson, 2022-07-18:
I was not able to compile code referencing SPI.h with this platformio.ini:

[env:nano_33_iot]
platform = atmelsam
board = nano_33_iot
framework = arduino
lib_deps =
    ; name-based (built-in library in framework)
    ; wifi nina 
    arduino-libraries/WiFiNINA@^1.8.13

Changing board to  mzeroUSB corrected the errors.  Is there a better way to correct this?

ivankravets, 2022-07-19:
Could you try to add `SPI` to the `lib_deps`?

jechiasson, 2022-07-19:
That fixed the compile error, thank you.  Why isn't it pulled in with the board?  SPI is required to communicate to the wifinina.

ivankravets, 2022-07-19:
https://github.com/arduino-libraries/WiFiNINA/blob/d0d58fd3017320938841e245da6bc8a7d77657a3/examples/AP_SimpleWebServer/AP_SimpleWebServer.ino#L19

ivankravets, 2022-07-19:
You can add `#include <SPI.h>` to your project source file and remove `SPI` from `lib_deps`. It will work too.