platformio/platformio-docs #281: When using esp-idf, libraries downloaded by PlatformIO are not found, but there is workaround

AgainPsychoX, 2023-01-03:
See https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components
(source at https://github.com/platformio/platformio-docs/blob/develop/frameworks/espidf_extra.rst)

By docs, `EXTRA_COMPONENT_DIRS` is to be set as library name. It would work if one had their library directly in main folder (which contains `src`, etc.). By some digging I found out, you can actually make it work better, by adding `.pio/libdeps/${board}` as components search directory, i.e.:

```cmake
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS .pio/libdeps/esp32cam)
project(YellowToyCar)
```

The solution might need some refining (and confirmation) I guess. I am not confident with PlatformIO / esp-idf myself, but I am posting it here (as an issue) so maybe other, more knowledgeable people could look at it. Anyway, it works for me ;)

valeros, 2023-01-03:
Hi @AgainPsychoX!

In a nutshell, this is expected behavior and I'd recommend storing your full-featured IDF components outside the `lib` folder to keep your project backward compatible with the native IDF build system. The typical approach is to create a new folder `components` in the root of your project and clone any dependency in the form of a Git submodule (take a look at the AWS-IoT example https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-aws-iot). This way you won't even need to set the `EXTRA_COMPONENT_DIRS` variable. Anyway, if you need a library from the PlatformIO Registry to be processed as an IDF component (e.g. reading options from the `Kconfig` file) your approach is just fine, as these libraries have to be added to the `EXTRA_COMPONENT_DIRS` one way or another. Just keep in mind that, there is a possible issue that these components may be built twice, once by the IDF build system and the second time by PlatformIO which is not what developers expect. Proper integration of the IDF components is tracked in the platform repository: https://github.com/platformio/platform-espressif32/issues/479 and https://github.com/platformio/platform-espressif32/issues/453.

closed 2023-01-03 by valeros

Jimis333, 2023-01-18:
Hi
If I get it right you make a "components" folder inside your root folder and place your libraries there. 

I'm trying to flash this example from Git 
https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/softAP

I started looking up the libraries needed and placing them inside the components folder in root. I soon realized that they are endless and some are more than one in many folders so you don't know which one to choose.
So I just copy-pasted the complete "components" folder from .. c:\Users\...\esp\esp-idf\components\ to my root. 
(Adding the path in C/C++ Extension configuration  ..\esp\esp-idf\components\ didn't work)

The intellisense was satisfied and had no warning about missing libs but 
1) when I try to build it using the command prompt ESP-IDF 5.0 CMD I get an error
" .../esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:128: undefined reference to `app_main'"
2) If I compile through VS code platformio I get another error shown in the picture

Can someone point to the right direction on solving this ?

[image]