platformio/platformio-docs #296: Critical link to Jenkins tutorial dead

maxgerhardt, 2023-03-24:
For "Integration", the Jenkins docs

https://docs.platformio.org/en/latest/integration/ci/jenkins.html#integration

point at 

http://www.thingforward.io/techblog/2017-11-03-setting-up-a-jenkins-ci-engine-for-embedded-projects.html

the domain however as ceased to exist apperently.

The WayBackMachine still has it archived though

https://web.archive.org/web/20200922231607/http://www.thingforward.io/techblog/2017-11-03-setting-up-a-jenkins-ci-engine-for-embedded-projects.html

vazhnov, 2023-04-07:
https://github.com/platformio/platformio-docs/blob/develop/integration/ci/jenkins.rst should be fixed or adjusted.
The original article at ThingForward blog wasn't opensource, so it can't just be copied into the PlatformIO documentation.
Somebody should write it from scratch…

ivankravets, 2023-04-08:
We temporarily removed that page. We would be thankful for the new PR.

commit e3c5bb85 (2023-04-08): Remove outdated docs for Jenkins CI // Resolve #296
closed 2023-04-08 by ivankravets