platformio/platformio-docs #313: PlatformIO Core Installer Script fail

teruyamato0731, 2023-08-12:
[Installer Script](https://docs.platformio.org/en/stable/core/installation/methods/installer-script.html#codecell0) fail.
```bash
python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
```
```log
curl: (22) The requested URL returned error: 404
```

Because [the required file has been deleted.](https://github.com/platformio/platformio-core/commit/89f4574680e6269e562bff849a90372a33ed3c40#diff-9c13707788b2dfd92c5c48ad6e55a75178f72f180836743f2ca3546104ebfef5)

So you need to restore the file or update the documentation.

ivankravets, 2023-08-14:
Solved in https://github.com/platformio/platformio-docs/commit/6b2af96f74d1c99f934a6975d9c26a4325abd172

closed 2023-08-14 by ivankravets

teruyamato0731, 2023-08-14:
This issue is still unresolved.

```bash
$ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py)"
# bash: /usr/bin/python3: Argument list too long
```

Running the command as above will result in an error, `Argument list too long`.
Note that my running environment is Ubuntu 22.04. Also, when I run `getconf ARG_MAX`, it returns ``2097152``.

There are two possible solutions: using standard input and pipes, or using bash process substitution.

Process substitution is as follows.
```bash
python3 <(curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py)
# Traceback (most recent call last):
#   File "/dev/fd/63", line 70, in <module>
#   File "/dev/fd/63", line 52, in main
#   File "/dev/fd/63", line 33, in create_temp_dir
#   File "/usr/lib/python3.10/tempfile.py", line 507, in mkdtemp
#     _os.mkdir(file, 0o700)
# FileNotFoundError: [Errno 2] No such file or directory: '/proc/3302/fd/.piocore-installer-jebo0myl'
```

Using standard input and pipes.
```bash
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py | python3
# Installer version: 1.2.1
# Platform: Linux-6.2.0-26-generic-x86_64-with-glibc2.35
# Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
# Python path: /usr/bin/python3
# Creating a virtual environment at /home/vscode/.platformio/penv
# Error: Could not find compatible Python 3.6 or above in your system.Please install the latest official Python 3 and restart installation:
# https://docs.platformio.org/page/faq.html#install-python-interpreter
```

As process substitution is not in Bourne Shell, it is better to use standard input and pipes.
In any case, both commands seem to fail.

A peek into the contents of the Installer Script shows that it references `__file__` in the script, and when process substitution is used, the tmpdir cannot be created in `__file__` because it does not have permissions, and an error occurs.

The cause of the error when using standard input and pipes was unknown.
It says `Could not find compatible Python`, but my environment has Python 3.10.6 installed as standard on Ubuntu 22.04.
Also, if I download the file and then run it, instead of using a one-liner, it succeeds.

```bash
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -o get-platformio.py
# python3 get-platformio.py
# Installer version: 1.2.1
# Platform: Linux-6.2.0-26-generic-x86_64-with-glibc2.35
# Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
# Python path: /usr/bin/python3
# Creating a virtual environment at /home/vscode/.platformio/penv
# Updating Python package manager (PIP) in a virtual environment
# PIP has been successfully updated!
# Virtual environment has been successfully created!
# Installing PlatformIO Core
# Collecting platformio
# ...
# Omitted below.
```

I am partially using machine translation as I am Japanese and my English is not good enough. I apologise if my English is inadequate.

I love Platform IO.
Thank you.

Original text as follows:

このissueはまだ解決されていません。

```bash
$ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py)"
# bash: /usr/bin/python3: Argument list too long
```

上記のようにコマンドを実行するとエラーが発生します。
なお、私の実行環境はUbuntu22.04です。また`getconf ARG_MAX`を実行すると`2097152`と返ってきます。

解決策として標準入力とパイプを使用する方法とbashのプロセス置換を使用する方法の2つが考えられます。

プロセス置換は以下の通りです。
```bash
python3 <(curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py)
# Traceback (most recent call last):
#   File "/dev/fd/63", line 70, in <module>
#   File "/dev/fd/63", line 52, in main
#   File "/dev/fd/63", line 33, in create_temp_dir
#   File "/usr/lib/python3.10/tempfile.py", line 507, in mkdtemp
#     _os.mkdir(file, 0o700)
# FileNotFoundError: [Errno 2] No such file or directory: '/proc/3302/fd/.piocore-installer-jebo0myl'
```

標準入力とパイプを用いた場合、
```bash
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py | python3
# Installer version: 1.2.1
# Platform: Linux-6.2.0-26-generic-x86_64-with-glibc2.35
# Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
# Python path: /usr/bin/python3
# Creating a virtual environment at /home/vscode/.platformio/penv
# Error: Could not find compatible Python 3.6 or above in your system.Please install the latest official Python 3 and restart installation:
# https://docs.platformio.org/page/faq.html#install-python-interpreter
```

プロセス置換はBourne Shellに無いため、標準入力とパイプを用いたほうが良いと思います。
いずれにせよどちらのコマンドも失敗するようです。

Installer Scriptの中身を覗いたところ、スクリプト内で`__file__`を参照しており、プロセス置換を用いた場合は権限がないため`__file__`内にtmpdirを作成できず、エラーが発生しているようです。

標準入力とパイプを用いた場合のエラーの原因は不明でした。
`Could not find compatible Python`と表示されますが私の環境にはUbuntu22.04に標準でインストールされるPython 3.10.6がインストールされています。
また、ワンライナーではなく、ファイルをダウンロードしてから実行した場合は成功します。
```bash
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -o get-platformio.py
# python3 get-platformio.py
# Installer version: 1.2.1
# Platform: Linux-6.2.0-26-generic-x86_64-with-glibc2.35
# Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
# Python path: /usr/bin/python3
# Creating a virtual environment at /home/vscode/.platformio/penv
# Updating Python package manager (PIP) in a virtual environment
# PIP has been successfully updated!
# Virtual environment has been successfully created!
# Installing PlatformIO Core
# Collecting platformio
# ...
# 以下省略
```

私は日本人であり英語が不得意なため、部分的に機械翻訳を使用しています。私の英語に不十分なところがあれば申し訳ありません。

I love Platform IO.
Thank you.

ivankravets, 2023-08-14:
We updated docs https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

Have you tried official methods?

ivankravets, 2023-08-14:
Thanks, fixed!

What is your use case for using the installer script?

teruyamato0731, 2023-08-14:
I use it in [the Dockerfile in this repo](https://github.com/teruyamato0731/pio-template/blob/main/.devcontainer/Dockerfile).

I want to use Platform IO in devcontainer and [CI (in devcontainer)](https://github.com/teruyamato0731/pio-template/blob/main/.github/workflows/main.yml).

ivankravets, 2023-08-14:
See updated docs https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

teruyamato0731, 2023-08-15:
Many thanks!
I updated [Dockerfile like this.](https://github.com/teruyamato0731/pio-template/compare/adf5413d02df0e71b42466ca00448a01131e8837...678318efe5327485ea552b0c065861cf6b3578c5)
And it worked on devcontainer and CI!

ivankravets, 2023-08-15:
Happy coding with PlatformIO! ;)