platformio/platformio-docs #326: PLatformIO's external scripts How to use them, middleware functions are a bit problematic

GPTKEY, 2023-11-09:
I want to replace the library files under the kernel package with files from the project Lib library.
I need to use an external script

I have repeatedly passed xxx.c directly to an external script using env.AddBuildMiddleware(replace_node_with_another, lib_file) and only see AddBuildMiddleware executed once at compile time, is this a bug?
The replace_node_with_another() function does not return None.

I've added extra_scripts to the ini file =
  pre:extra_script.py

valeros, 2023-11-09:
Hi @GPTKEY, do you mind sharing your `platformio.ini` and the code from the extra script here?

GPTKEY, 2023-11-10:
[env:bluepill_f103c8]
board = bluepill_f103c8
platform = ststm32
framework = arduino
extra_scripts = pre:extra_script.py
build_flags = -I./lib/uart/
lib_deps = 
	bxparks/AceButton@^1.9.2
	arduino-libraries/ArduinoRS485@^1.0.2
	arduino-libraries/ArduinoModbus@^1.0.7
	4-20ma/ModbusMaster@^2.0.1
	vermut/Modbus@^1.0.0
	yaacov/ModbusSlave@^2.1.1
debug_tool = jlink
upload_protocol = jlink
; platform_packages = framework-arduinoststm32

middleware_functions = [
    "**/uart.c",
    "**/uart.h"
]

for file_path in middleware_functions :
    env.AddBuildMiddleware(replace_node_with_another, file_path)
    
I tested it many times and finally implemented AddBuildMiddleware() to run successfully.
But it didn't work the way I wanted it to when I compiled it later.
Because when I replace the header file, it doesn't work.
Maybe the header file is not supported, or there are some parameter variations that prevent the header file from being fully replaced.

valeros, 2023-11-10:
Unfortunately, only source files can be used with `AddBuildMiddleware` helper. Not sure if it's applicable in your case, but there is a possible alternative for overriding default packages described here https://docs.platformio.org/en/latest/scripting/examples/override_package_files.html

closed 2023-11-10 by valeros