platformio/platformio-docs #348: Setting monitor_filters in ini file as described in the documentation does not work

marcelloDC, 2024-04-23:
When I set `monitor_filters` to `direct` using the  multiple lines format as described in the documentation I get the following warning when i start up the serial-monitor:

`Warning! Skipping unknown filters direct; . Known filters are colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time`

and the filter is not set. It only works when I use the "single line" format as below:

`monitor_filters = direct`

ivankravets, 2024-04-24:
> Warning! Skipping unknown filters direct; 

Wrong declaration. The format is following:

```
monitor_filters = 
   f1
   f2
   fn
```

closed 2024-04-24 by ivankravets

ivankravets, 2024-04-24:
Indeed, the `;` is a valid symbol, this is a comment identifier. However, it should be split with `space`. For example,
```ini
monitor_filters =
    direct ; my comment
```