platformio/platformio-docs #356: Update suggestion for Arduino UNO R4 Wifi

atrantan, 2024-06-11:
Hi all,
may I suggest to update the documentation page of Arduino Uno R4 Wifi since it is now '1-click' debuggable with no extra external debug probe here  [https://github.com/platformio/platformio-docs/blob/develop/boards/renesas-ra/uno_r4_wifi.rst](url)
My `platformio.ini` looks like below
```txt
[env:uno_r4_wifi]
platform = renesas-ra
board = uno_r4_wifi
framework = arduino
debug_tool = cmsis-dap
upload_protocol = sam-ba
```
Just one issue though. When I'm opening a debugging session, everything works well (breakpoints, step over, etc.). But for the next debug session / upload, I need to unplug/plug the USB cable since I get the following error
in gdb
```txt
no device found on COM xx
```

atrantan, 2024-06-11:
Just another remark : debug sessions work perfectly with Arduino IDE, the 'no device found' issue should be easily fixed here [https://github.com/platformio/platform-renesas-ra](url)

valeros, 2024-06-11:
Hi @atrantan, the board pages are autogenerated according to board manifest files in the platform repository. I've fixed the issue there, so the docs will be updated when a new version of the Renesas dev-platform is released.

closed 2024-06-11 by valeros